ADVRHumanoids/XBotCore #5: Shared memory issue with std::any

pepperig, 2017-11-01:
@alaurenzi I am creating this issue so we do not forget. 
We have a SEGFAULT on closing phase when we use the shared memory. The problem in on the destructor of the std::any(). NOTE that std::any allocate the shared_ptr on the heap.

alaurenzi, 2017-11-02:
Before starting with a possible refactor of the `SharedMemory` (which anyway will have to be done to make it thread-safe), I would like to make sure that putting shared pointers inside a `boost::any` is actually causing problems. Can you verify this with a small self-contained example?

pepperig, 2017-11-02:
When possible I can try. However in debug mode you can see the error.

alaurenzi, 2017-11-03:
Refactored in 0f6b35872dbdefcc4d30ba8, apparently the closing is now ok

liesrock, 2017-11-05:
Great, I close it.

closed 2017-11-05 by liesrock