ADVRHumanoids/XBotCore #7: WIKI

liesrock, 2017-11-16:
Discussion about the wiki contents and strucutre.

alaurenzi, 2017-11-16:
As a first step, let us come out with a general structure. Then we can create a skeleton and start filling it in parallel!

liesrock, 2017-11-16:
I would use https://armarx.humanoids.kit.edu/index.html as a starting point: what do you think?

nkashiri, 2017-11-16:
It looks, but it has some FEATURES that I hope we will not have, for instance, in the main documentation "ArmarXDoc", there is an "extended documentation" (that would be nice if we have in the main page), in which there is another "documentation" page! 
I would say it would be nice if we could have that "extended documentation" integrated to the main page. 
From a user perspective, like me, this documentation is the ideal case: (very nice index list)
http://www.coppeliarobotics.com/helpFiles/index.html

liesrock, 2017-12-04:
Ok given your feedbacks I started creating the skeleton of the WIKI: I am currently offline working on the config files section

https://github.com/ADVRHumanoids/XBotCore/wiki

alaurenzi, 2017-12-09:
I'm almost finished with the XbotInterface page (https://github.com/ADVRHumanoids/XBotCore/wiki/XBotInterface). I will also take care of the ROS integration part.

Take a look and give feedback @liesrock  @pepperig  @nkashiri  
PS: Whoever wants to review the english is welcome :D

pepperig, 2017-12-11:
it looks very good

MartijnZ, 2017-12-11:
Hi,

I did an update on the English of the interface documentation. But I don't know how to apply my changes (I can't push).

Martijn

liesrock, 2017-12-19:
I have completed the configuration files section (https://github.com/ADVRHumanoids/XBotCore/wiki/Configuration-files) and revised the quick start and the XBotInterface part. Super nice work!
I added a new section (XBotCore) where we are going to describe in deep the plugins (RT, IO, NRT) and the XBotCore requirements (in terms of URDF/SRDF constraints etc.)..
Please when you have time revise and give feedback.

alaurenzi, 2017-12-21:
Started with the [XbotCore + ROS part](https://github.com/ADVRHumanoids/XBotCore/wiki/XbotCore-ROS-integration)