ADVRHumanoids/XBotCore #8: Control mode from command line

alaurenzi, 2017-11-21:
How easy is it to set the robot to a standard control mode from command line?   
Example:
- `XBotCore --idle`
- `XBotCore --torque`
- `XBotCore --pos`

@liesrock @pepperig

pepperig, 2017-11-21:
I asked time ago to @liesrock  but the problem is that each slave can be controlled in different mode..so a general mode implies the same mode for every slave.

alaurenzi, 2017-11-21:
Yes! We can define a set of simple rules like:
 - a wheel only supports velocity and idle
 - a low power esc only supports position and idle

pepperig, 2017-11-21:
ok! what about having joint partially controlled in torque mode? or we do not consider this case..

alaurenzi, 2017-11-21:
Meaning?

pepperig, 2017-11-21:
For example having the shoulder in position and the arm in torque control

liesrock, 2018-03-01:
@alaurenzi can you better explain what you would like to have from this command line tool?

liesrock, 2018-04-17:
I think @pepperig can now provide this tool with the hal2.0