ADVRHumanoids/XBotCore #10: Dealing with NED IMU orientation

alaurenzi, 2017-12-07:
Recently we found out with Luca that our IMU will provide the orientation of its own sensor frame (as depicted on the sensor itself) and a world frame according to the NED (north-east-down) convention. Since we work with a world with +Z pointing upwards, we need to take care of this fact appropriately.

My suggestion is to compensate for such a difference by pre-applying a rotation of +180 degrees about the X axis when reading from the sensor, i.e. at the `XBotEcat` layer.

This is equivalent to using the following code for reading the IMU orientation:
```

liesrock, 2018-03-01:
We currently use the IMU with @alaurenzi modification.

alaurenzi, 2018-03-01:
I reopen since if I remember correctly that one is just a local change on centauro embedded.. not really a solution!

liesrock, 2018-03-01:
Ok, thanks.
The issues can't be here forever: I ask the others mentioned to give us their opinion ASAP and to find a real solution.

@nkashiri @ntsagarakis @liesrock @EnricoMingo @pepperig

liesrock, 2018-04-12:
Ok, we take this as the final solution to solve the above described issue: I will merge in this https://github.com/ADVRHumanoids/XBotEcat/pull/1

@ADVRHumanoids/advr-developers

closed 2018-04-12 by liesrock