ADVRHumanoids/XBotCore #16: robot_state_publisher instance inside CommunicationInterfaceROS causes segfault

liesrock, 2018-01-19:
As you are already aware, on a clean 16.04 + xeno 3 installation I have a segfault on the instantiation of a ROS robot_state_publisher.
It is not easy to debug the error, but removing the robot state publisher for the core can be an option as (@EnricoMingo) was mentioning.

@alaurenzi @pepperig ideas?

alaurenzi, 2018-01-19:
I would not like to sacrifice the usability of the framework (only one terminal to use it). We can use an external rs publisher only if we manage to launch everything from one terminal (e.g. roslaunch).

We really should understand what is behind this very strange segmentation fault, because it will happen again for sure, and maybe the next time it will affect a vital component of the system.

alaurenzi, 2018-01-25:
If I understand correctly, in the latest master we don't have the integrated RS publisher anymore. Is this correct?

liesrock, 2018-04-17:
@alaurenzi can you update us in this?

alaurenzi, 2018-04-17:
Latest XBotCore devel contains an internal robot state publisher that looks to solve the problem

liesrock, 2018-06-11:
I confirm that in this way the problem is solved: ok in devel I will merge it in master ASAP

closed 2018-06-11 by liesrock