ADVRHumanoids/XBotCore #17: XBotCore fails to load libXBotEcat.so

DaniloCaporale, 2018-01-22:
```
XBotCore configs/ADVR_shared/user_example/walkman_debris_simulation.yaml
[info] XBotCore using config file /home/danilo/projects/advr-superbuild/configs/ADVR_shared/user_example/walkman_debris_simulation.yaml
[error] ETHERCAT HAL interface NOT found! 
/home/danilo/projects/advr-superbuild/build/install/lib/libXBotEcat.so: cannot open shared object file: No such file or directory
```

Same goes with `walkman_example.yaml`.

Am I missing some package? Tried with XBotEcat but it doesn't compile on my machine.

pepperig, 2018-01-22:
Please check the wiki at https://github.com/ADVRHumanoids/XBotCore/wiki/quick-start.
Try using the dummy option -> XBotCore -D

DaniloCaporale, 2018-01-22:
`XBotCore -D configs/ADVR_shared/user_example/walkman_example.yaml` works! Thanks!

closed 2018-01-22 by pepperig