ADVRHumanoids/XBotCore #19: Run simulation in gazebo

DaniloCaporale, 2018-01-24:
Read the docs alrady, can't figure how what I'm missing.

I do:
```
roscore
XBotCore -D walkman_example.yaml
```
and have to use -D otherwise I find #17, but in this case I got kinematics only.
Infact, if I run `gazebo --verbose` and this is the output I get:

```
[Err] [InsertModelWidget.cc:298] Missing model.config for model "/home/danilo/projects/advr-superbuild/robots/iit-walkman-ros-pkg/walkman_gazebo/database/walkman_urdf"
[Err] [InsertModelWidget.cc:298] Missing model.config for model "/home/danilo/projects/advr-superbuild/robots/walkman_final_demo_field/.git"
```

Then adding the robot I find:

```
[Wrn] [msgs.cc:1655] Conversion of sensor type[imu] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Err] [Model.cc:1010] Exception occured in the Load function of plugin with name[xbot_plugin_handler] and filename[libGazeboXBotPlugin.so]. This plugin will not run.
```

and the robot falls down.
@liesrock @pepperig any ideas?
Thanks!!!

liesrock, 2018-01-24:
Ok, but I ask you again to read the docs, because this is not the way to run the simulation in verbose mode as stated here:

https://github.com/ADVRHumanoids/XBotCore/wiki/configuration-files

```
SimulationOptions:
  verbose_mode: "false"
```
Node SimulationOptions (Optional)
verbose_mode (Mandatory) -> true if you want the Gazebo simulation with verbose command line output, false otherwise.

Moreover the options to pass to XBotCore are written here:

https://github.com/ADVRHumanoids/XBotCore/wiki/quick-start

```
user@user-desktop:~$ XBotCore --help
XBotCore. Available options:
  -C [ --config ] arg   Path to custom config file. If not set, a default 
                        config file must be configured via set_xbot_config.
  --ch-config arg       Path to custom config file for the 
                        CommunicationHandler. If not set, the same of XBotCore 
                        is used.
  --no-ch               Do not run the CommunicationHandler
  --dont-catch          Disable catching exceptions thrown inside rt plugins 
                        init()
  --period arg          XBotCore thread period [us]
  --ch-period arg       CommunicationHandler thread period [us]
  -D [ --dummy ]        Use the dummy HAL implementation.
  -V [ --verbose ]      Verbose mode.
  --help                Shows this help message.
```

In the quick start you select the config file to use using the set_xbot_config, so you don't have to pass again the config file to XBotCore exe.

Let's start with the kinematic simulation and then we check gazebo; just run:

```
roscore
XBotCore -DV
```

and check the robot on RViz.

DaniloCaporale, 2018-01-24:
```
XBotCore -C /home/danilo/projects/advr-superbuild/configs/ADVR_shared/user_example/walkman_simple_example.yaml -DV
rviz
```
In rviz I add a RobotModel, select base_link as fixed frame, put `xbotcore/walkman/robot_description` in Robot Description, among the RobotModel options, and I see the robot.
Then:
```
rosservice call /HomingExample_switch "data: true" 
```
and the robot does homing properly in rviz.

What about gazebo?

liesrock, 2018-01-24:
Super!

Now let's try to update your config file to have a verbose mode in gazebo with 

```
SimulationOptions:
  verbose_mode: "true"
```

and lunch:

```
roscore
gazebo
```

and drag and drop the walkman robot.

Then copy paste the output and we can analyze the error J

DaniloCaporale, 2018-01-24:
```
Gazebo multi-robot simulator, version 7.0.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 7.0.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 131.114.31.64
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 131.114.31.64
[Err] [InsertModelWidget.cc:298] Missing model.config for model "/home/danilo/projects/advr-superbuild/robots/iit-walkman-ros-pkg/walkman_gazebo/database/walkman_urdf"
[Err] [InsertModelWidget.cc:298] Missing model.config for model "/home/danilo/projects/advr-superbuild/robots/walkman_final_demo_field/.git"
[Wrn] [msgs.cc:1655] Conversion of sensor type[imu] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Err] [Model.cc:1010] Exception occured in the Load function of plugin with name[xbot_plugin_handler] and filename[libGazeboXBotPlugin.so]. This plugin will not run.

----------------------------------------------------------------------------------
----------------------------------------------------------------------------------
Successfully load gazebo_magnetic_attach_plugin for: walkman::LSoftHand

---------------------------------------
*** How to: ***
---------------------------------------
To grasp an object: 
	$ rosservice call /grasp/LSoftHand "grasp: 1"

To detach the object: 
	$ rosservice call /grasp/LSoftHand "grasp: 0"

Service response: success (1 or 0)!

Note: 
  Make sure you have touched at least one object! 
  The attachment will be established at current distance! 
----------------------------------------------------------------------------------
----------------------------------------------------------------------------------

----------------------------------------------------------------------------------
----------------------------------------------------------------------------------
Successfully load gazebo_magnetic_attach_plugin for: walkman::RSoftHand

---------------------------------------
*** How to: ***
---------------------------------------
To grasp an object: 
	$ rosservice call /grasp/RSoftHand "grasp: 1"

To detach the object: 
	$ rosservice call /grasp/RSoftHand "grasp: 0"

Service response: success (1 or 0)!

Note: 
  Make sure you have touched at least one object! 
  The attachment will be established at current distance! 
----------------------------------------------------------------------------------
----------------------------------------------------------------------------------

[Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/physics/contacts, deleting message. This warning is printed only once.

```

and the robot is KO.

liesrock, 2018-01-24:
Hi @DaniloCaporale this is how the verbose output should look like:

```
lucamuratore@lucamuratore-desktop:~$ gazebo
[info] getXBotConfig -> /home/lucamuratore/advr-superbuild/configs/ADVR_shared/user_example/walkman_debris.yaml
[info] configs/ADVR_shared/walkman/configs/walkman.yaml
[info] /home/lucamuratore/advr-superbuild/configs/ADVR_shared/walkman/configs/walkman.yaml
[info] Control rate is 0.001
[info] Set RosHandle! 0x7f0db3016c50
[info] configs/ADVR_shared/walkman/configs/walkman.yaml
[info] /home/lucamuratore/advr-superbuild/configs/ADVR_shared/walkman/configs/walkman.yaml
[info] Joint LHipLat gains -> P = 8000, D = 30
[info] Joint # 0 - LHipLat
[info] Joint LHipYaw gains -> P = 3000, D = 10
[info] Joint # 1 - LHipYaw
[info] Joint LHipSag gains -> P = 8000, D = 30
[info] Joint # 2 - LHipSag
[info] Joint LKneeSag gains -> P = 8000, D = 30
[info] Joint # 3 - LKneeSag
[info] Joint LAnkSag gains -> P = 8000, D = 30
[info] Joint # 4 - LAnkSag
[info] Joint LAnkLat gains -> P = 8000, D = 30
[info] Joint # 5 - LAnkLat
[info] Joint RHipLat gains -> P = 8000, D = 30
[info] Joint # 6 - RHipLat
[info] Joint RHipYaw gains -> P = 3000, D = 10
[info] Joint # 7 - RHipYaw
[info] Joint RHipSag gains -> P = 8000, D = 30
[info] Joint # 8 - RHipSag
[info] Joint RKneeSag gains -> P = 8000, D = 30
[info] Joint # 9 - RKneeSag
[info] Joint RAnkSag gains -> P = 8000, D = 30
[info] Joint # 10 - RAnkSag
[info] Joint RAnkLat gains -> P = 8000, D = 30
[info] Joint # 11 - RAnkLat
[info] Joint WaistLat gains -> P = 8000, D = 30
[info] Joint # 12 - WaistLat
[info] Joint WaistYaw gains -> P = 8000, D = 30
[info] Joint # 13 - WaistYaw
[info] Joint LShSag gains -> P = 4000, D = 10
[info] Joint # 14 - LShSag
[info] Joint LShLat gains -> P = 3000, D = 10
[info] Joint # 15 - LShLat
[info] Joint LShYaw gains -> P = 3000, D = 10
[info] Joint # 16 - LShYaw
[info] Joint LElbj gains -> P = 3000, D = 10
[info] Joint # 17 - LElbj
[info] Joint LForearmPlate gains -> P = 300, D = 1
[info] Joint # 18 - LForearmPlate
[info] Joint LWrj1 gains -> P = 1500, D = 10
[info] Joint # 19 - LWrj1
[info] Joint LWrj2 gains -> P = 300, D = 1
[info] Joint # 20 - LWrj2
[info] Joint NeckYawj gains -> P = 300, D = 1
[info] Joint # 21 - NeckYawj
[info] Joint NeckPitchj gains -> P = 300, D = 1
[info] Joint # 22 - NeckPitchj
[info] Joint RShSag gains -> P = 4000, D = 10
[info] Joint # 23 - RShSag
[info] Joint RShLat gains -> P = 3000, D = 10
[info] Joint # 24 - RShLat
[info] Joint RShYaw gains -> P = 3000, D = 10
[info] Joint # 25 - RShYaw
[info] Joint RElbj gains -> P = 3000, D = 10
[info] Joint # 26 - RElbj
[info] Joint RForearmPlate gains -> P = 300, D = 1
[info] Joint # 27 - RForearmPlate
[info] Joint RWrj1 gains -> P = 1500, D = 10
[info] Joint # 28 - RWrj1
[info] Joint RWrj2 gains -> P = 300, D = 1
[... 1158 lines cut ...]
[info] Advertised service OpenSotIk_switch
[info] Advertised service IkRosSMPub_switch
[info] Advertised service Grasp_switch
[info] Advertised service XBotCommunicationPlugin_switch
[info] Advertised service HomingExample_cmd
[info] Advertised service OpenSotIk_cmd
[info] Advertised service IkRosSMPub_cmd
[info] Advertised service Grasp_cmd
[info] Advertised service XBotCommunicationPlugin_cmd
[info] Advertised status port for plugin HomingExample
[info] Advertised status port for plugin OpenSotIk
[info] Advertised status port for plugin IkRosSMPub
[info] Advertised status port for plugin Grasp
[info] Advertised status port for plugin XBotCommunicationPlugin
[info] THREAD INIT: name = XBOT_COMMHANDLER, period 5000 us
[success] Thread XBOT_COMMHANDLER: start looping
```

So in your case it looks like the plugin is not starting at all: can you check which config file you are using by typing:

```
get_xbot_config
```

And can you double check that the plugin is correctly installed and the $GAZEBO_PLUGIN_PATH correctly set?

DaniloCaporale, 2018-01-24:
Used `set_xbot_config` to get the correct path. Works better now.
`$GAZEBO_PLUGIN_PATH ` looks correct, I can see e.g.
```
libDefaultGazeboPID.so  libGazeboImpl.so  libGazeboXBotPlugin.so
```
in that folder.
Still, gazebo doesn't start properly:
```
[Err] [InsertModelWidget.cc:298] Missing model.config for model "/home/danilo/projects/advr-superbuild/robots/iit-walkman-ros-pkg/walkman_gazebo/database/walkman_urdf"
[Err] [InsertModelWidget.cc:298] Missing model.config for model "/home/danilo/projects/advr-superbuild/robots/walkman_final_demo_field/.git"
```
and when I add the robot:
```
[Wrn] [msgs.cc:1655] Conversion of sensor type[imu] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.
[info] getXBotConfig -> /home/danilo/projects/advr-superbuild/configs/ADVR_shared/walkman/configs/walkman.yaml
[info] Control rate is 0.001
[info] Set RosHandle! 0x7fb038be0770
ERROR in parseYAML : YAML file  /home/danilo/projects/advr-superbuild/configs/ADVR_shared/walkman/configs/walkman.yaml  does not contain XBotInterface mandatory node!!
[Err] [Model.cc:1010] Exception occured in the Load function of plugin with name[xbot_plugin_handler] and filename[libGazeboXBotPlugin.so]. This plugin will not run.
```

Am I using the correct yaml files to run simulation in gazebo?

liesrock, 2018-01-24:
Great! Yes the issue was the config file you were using: now that you are confident with the set_xbot_config I suggest to use the following config file:
```
https://github.com/ADVRHumanoids/ADVR_shared/blob/master/user_example/walkman_simple_example.yaml
```
pull the latest XBotCore and ADVR_shared and you will have everything working! Then I suggest to speak with @rsubburaman to understand which config file he is using for the DoorOpening task.

liesrock, 2018-01-25:
Any news on this @DaniloCaporale ?

DaniloCaporale, 2018-01-25:
Hi there,

just pulled everything and set the correct xbot_config file. I get this error when starting xbotcore
```
XBotCore -DV
...
[success] Successfully initialized AdvrJointState message interface!
[success] Send robot state over ROS ok!
[success] WEB_SERVER INTERFACE found! 
[info] XBotCore server running at http://127.0.0.1:8081
[error] JointStateRemapper IO plugin NOT found! 
/home/danilo/projects/advr-superbuild/build/install/lib/libJointStateRemapper.so: cannot open shared object file: No such file or directory
[info] Initializing IO plugin JointStateRemapper...
Segmentation fault (core dumped)
```

I should still use the `-DV` option right?

DaniloCaporale, 2018-01-25:
Can't find `libJointStateRemapper.so` in my system. Am I missing some package?

liesrock, 2018-01-25:
Pull rhe latest XBotCore master!

DaniloCaporale, 2018-01-25:
Already up to date

liesrock, 2018-01-25:
It's here: e9d21742c4ed21174b39ea13d904c18d07068f76 - check if you successfully compiled

DaniloCaporale, 2018-01-25:
I was already there. Rebuilt but it's still the same.

liesrock, 2018-01-25:
Strange, that commit add the plugin: to try remove it from the config (you are not going to have the TF)

DaniloCaporale, 2018-01-25:
Removing the plugin `XBotCore -DV` does not crash. Still, if I run gazebo and add the robot I get
```
[info] getXBotConfig -> /home/danilo/projects/advr-superbuild/configs/ADVR_shared/user_example/walkman_simple_example.yaml
[warning] in th_init! XBotRTPlugins node does NOT contain mandatory node io_plugins!
```
and then gazebo crashes.

liesrock, 2018-01-25:
By removing the plugin I mean remove only the "JointStateRemapper" not the whole YAML node that is mandatory:
```
io_plugins: []
```

DaniloCaporale, 2018-01-25:
Sorry :P
Removed correctly and now I get this when I add the robot, and gazebo crashes:

```
danilo@walkmanbox:~/projects/advr-superbuild/configs/ADVR_shared/user_example (master)$ gazebo
[info] getXBotConfig -> /home/danilo/projects/advr-superbuild/configs/ADVR_shared/user_example/walkman_simple_example.yaml
[info] Opening /tmp/danilo/Motor_id_26...
[info] Opening /tmp/danilo/rt_in_Motor_id_55...
[ WARN] [1516899089.329037399]: Converting unknown joint type of joint 'reference' into a fixed joint
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
```

DaniloCaporale, 2018-01-30:
Guys, we managed to make it work.
Just as a reference: we pulled everything new, then we did the following.

The correct sequence to start the simulation in gazebo is:

```
roscore
gazebo
```
then insert the robot within gazebo.

We could not find this stuff on the XBotCore wiki so I added it in a new page, hope it's useful.

closed 2018-01-30 by DaniloCaporale

liesrock, 2018-02-02:
merged here: https://github.com/ADVRHumanoids/XBotCore/wiki/Quick-Start