ADVRHumanoids/XBotCore #21: Impedance Ankle transmission plugin should always work on Cogimon

FrancescoRuscelli, 2018-02-09:
The ImpedanceAnkle plugin, which implement stiffness and damping compliance on the ankle joint, starts only when a move is issued.
Instead, it should always be running independently from the user issuing a move or not, as long as the robot is turned on.

alaurenzi, 2018-02-09:
I agree. Probably a good solution is to call transmission plugins directly inside the plugin handler loop... Let's discuss about this

liesrock, 2018-02-09:
Thanks @FrancescoRuscelli, I agree too: it was a bad design choice (documented here https://github.com/ADVRHumanoids/RobotInterfaceXBotRT/issues/3) -> I will keep both the issue updated.

alaurenzi, 2018-03-06:
This is quite relevant bug, ping!

pepperig, 2018-03-06:
I think @liesrock  is working on it

liesrock, 2018-03-06:
Yes, I am on it