ADVRHumanoids/XBotCore #24: Run time information about the HAL

pepperig, 2018-02-24:
Provides inside a plugin a way to check what HAL is loaded so to have different behavior in simulation or even in different robot. The goal is to avoid IFDEF forcing to recompile the code.

pepperig, 2018-03-01:
Done in commit 6e8c6e055e2e06e3985c4fe5564452724ce2ff04
@ADVRHumanoids/all  
You can find inside the HomingExample how to get the information to know if you are in simulation, debug(dummy mode) or in the real robot. You can also get info on the type of the robot.

liesrock, 2018-03-01:
Great! Can you please add all the information on the documentation page of the R-HAL?

https://github.com/ADVRHumanoids/XBotCore/wiki/R-HAL

We need to keep the documentation synchronized with the new features.

pepperig, 2018-03-01:
done!

closed 2018-03-01 by pepperig