ADVRHumanoids/XBotCore #28: Init plugin called from a not real time thread

pepperig, 2018-03-04:
I am not sure that breaking the RT even in the init (so for small amount of time) it is a good thing. I started think about it because currently the kuka robot implementation runs on the xbotcore thread. I realized that I am not able anymore to run at 5 ms because of the switching to secondary mode caused by the init of homing example. Anyway I am planning to have a separated thread for the robot.
However a solution could be to have the Loader thread handling the initialization of the plugins leaving the RT thread to handle the control loop.
Opinions? @liesrock @alaurenzi @EnricoMingo

liesrock, 2018-03-04:
I agree: in general the HAL (as we discussed) should give the possibility to use a NRT thread for this kind of operation. I am not sure how much we gain using the loader.

Anyhow this would lead to a basic refactor of the HAL interface right? 
What would we need to change in the current concrete implementations?
How are we going to handle the operations btw the RT thread and the NRT thread? Is this responsibility of the framework or of the developer of the HAL concrete implementation?

pepperig, 2018-03-04:
The idea is to have a separate RT thread that handles the robot (like the cat master thread).
We could think to provide both a RT and NRT HAL, but if possible I would focus on making RT safe the kaka implementation.
So yes, we have to change a bit the interface.
The communication will use iddp, xddp, shared memory depending on the type(RT, NRT) of HAL thread.

When I mentioned the init I was referring to the init method of each xbot plugin. Currently it is called inside the plugin handler by the RT threads. The current homing use the ros utils real time feature that cause context switch. We could think to have the loader (NRT) that has already a pointer to the plugin handler the job to call the init. In this way we do not break the RT.

pepperig, 2018-03-05:
Could be also related to this issue https://github.com/ADVRHumanoids/XCM/issues/54

pepperig, 2018-03-06:
#31

pepperig, 2018-03-15:
solved in #34

closed 2018-03-15 by pepperig