ADVRHumanoids/XBotCore #29: Make execution time plugin constant and predictable

pepperig, 2018-03-04:
So far we have a RT threads that spins at a given control period but we do not take into account the fact that a plugin could take longer execution time. We could have a sort of optional hard RT constraint that check if we exceed the time. This will be also useful for debug purpose.
What do you think? @liesrock @alaurenzi @EnricoMingo

liesrock, 2018-03-04:
The idea is cool, but what would you do if the execution time exceeds the hard RT constraint deadline? It is not easy to assure hard RT timing in a complex plugin, how would you check it? Only experimentally? 
Try to explain better what tool you would like to develop and how it should work.

pepperig, 2018-03-04:
What to do if the execution time exceed is not clear yet. We could just stop the execution of the control loop. Basically we could use it with the dummy mode to check beforehand the execution time.
Of course this will be an optional feature when we want have hard RT constraint. 
I would check the execution time of the plugin handler control loop rather that the plugin one.
The solution will use a timer.

pepperig, 2018-03-04:
Interesting point https://stackoverflow.com/questions/45958888/what-happens-if-a-hard-realtime-task-exceeds-its-deadline
So basically is up to us, but we must know the capability of the system and relax the period accordingly or optimize the code if we do not meet the deadline.

EnricoMingo, 2018-03-04:
Hi all,
If the plugin handler exceeds the control period, Xenomai gives an error. Of course it is interesting to understand which is the cause, that in general could be that the sum of the periods of each plugin (which is maybe under the ms) is higher than the control period. If a single plugin execution is over the control period is easy, the former case is more complicated because who is the plugin which should take less, in general? It is difficult to say in my opinion...I think that the good practice in this case is to allow more time for the control period.

Inviato da iPhone

pepperig, 2018-03-05:
I am not so sure that xenomai gives error if we exceed the control period because we have some log showing spike in the period. In my opinion it is good to know as first thing whether the global control loop time is under the control period. 
As you said checking the control period for each plugin is not easy. We could give a fixed amount of time to each plugin, so (period/number_of_plugin) but again it is not a fair policy because it depends on the amount of code of the plugin.

pepperig, 2018-03-06:
#31 gives waring on time control exceed. Not clear yet what to do if we exceed the time.

pepperig, 2018-03-09:
Just to recap a bit my understanding. Period = 1 means not period thread, otherwise if >1 it will be a periodic at the period's frequency. Just in a periodic thread we have the xenomai warning when the execution time exceeds the period.
Currently when we run XBotCore we spawn a non periodic thread which frequency depends on the Ecat master thread. 
The Ecat thread is independent from XBotCore, it spins at a given frequency and at each loop send a signal to the XBotCore thread to wake it up.
The XBotCore thread at each loop sleeps on the read_from_slave waiting for the Ecat thread to wake up.
The worst case in synchronization is when the XBotCore thread performs a reading (so goes to sleep) an instant after that the Ecat thread has done the signal to wake up XBotCore thread.
This means that XBotCore thread will sleep less (period - previous exceeded time).
Next loop will be fine and after we go again if the delay is constant.

pepperig, 2018-03-09:
[image]

This shows the worst case

liesrock, 2018-03-27:
We have more answers: the EtherCAT slave implementation (soes) on the firmware of the boards we are currently using requires in the worst case 400 microseconds to read and write a new frame on the PDO.
Given the current implementation of the multi-thread synchronization (ecat thread periodic @ 1ms and XBot thread non-periodic following ecat thread based on a condition variable), the recv_from_slaves function requires at least 400 microseconds, leading for us to a maximum control_loop time ~600 microseconds to stay inside the 1 milliseconds. Given that we have a set of operation running in the XBot which require 200 microsecond, the time left for a control plugin is just 400 microseconds.
I suggest to switch to 2 millisecond control waiting for another communication mechanism on the low level, or on investigating a different synchronization method.

pepperig, 2018-03-27:
@ntsagarakis

nkashiri, 2018-03-27:
I hope it's just for some experiments, and the plan is not to actually lower the frequency of inner most centralized controller to 500Hz. In 2018, it'll be a tad embarrassing to say it out of the lab..
Besides, for the controllers that are so necessary in RT (like impedance controller), it's still possible to survive with 0.4ms. 
In any case, @liesrock @alessiomargan , it would be nice to improve the implementation of ECAT and XBOT to give more space to the RT controller

pepperig, 2018-03-27:
I would suggest to implement a different synch mechanism. Each thread spins on its own.

liesrock, 2018-03-27:
@nkashiri the bottleneck is the SPI communications between the microcontroller and the beckhoff chip. I know Alessio will work on a parallel mechanism, meanwhile we can try what Giuseppe suggests.
Anyhow as you said, RT plugin runnable in less than 400 microsecond will be run at 1kHz, so no worries!

liesrock, 2018-03-29:
Just to add that @pepperig used the XBot on a KUKA controlled in RT mode and the overhead from the low level is negligible.
So the XBot can be use at 1 kHz with 200 microseconds overhead, keeping in mind that 100 microseconds of them are used to perform the sense of the robot and the update of the internal model, which all the plugins can use "for free" in the control loop.

alessiomargan, 2018-03-29:
[image]

the blue scope measure the ecat irq handler execution time, so tx and rx pdo takes 200us on the microcontroller
I am doing some measurements to understand others timing delay in the master.

alessiomargan, 2018-03-29:
Seems that an ecat_cycle(), send and receive,  in the ecat_thread is 350 us