ADVRHumanoids/XBotCore #39: Distributed Logger

pepperig, 2018-03-30:
As we all know the the actual RT logger is time limited. We can allocate bigger buffer but we are constrained by the system memory.
A solution could be to have the NRT thread dealing with the copy and transfer of the RT logging to the file. We will use XDDP pipe to forward data.

alaurenzi, 2018-04-02:
This is a welcome improvement. We should first understand how to use the matio library in order to modify an already existing mat file in order to add new data. Then we should find a way to share data with a nrt thread such that:
- the rt thread never blocks
- the rt thread performance is not affected

Logging is very common in our plugins and we don’t want any performance penalty on this!

liesrock, 2018-04-02:
What about using the ROS based robot state logger?