ADVRHumanoids/XBotCore #40: Cogimon ankles nrt control bug

alaurenzi, 2018-04-02:
When switching to NRT control, the ankles move without reason. Appare tly, they go to the motor-side zero.
We should work on solving this with high priority, otherwise it will be difficult to do the opensot demo for the upcoming review meeting.

@liesrock @pepperig

alaurenzi, 2018-04-02:
@EnricoMingo

liesrock, 2018-04-02:
I know about the issue but I did not give high priority to this since we always used the robot in RT mode: if you don't have the code ready to work as a RT plugin, and you don't want to deal with it, I will work on the transmission plugin issue starting from tomorrow afternoon.

liesrock, 2018-04-03:
OK solved with https://github.com/ADVRHumanoids/cogimon_tools/commit/8857abdb22deff7215818730571d7a840b4459c5

The reason why we had the issue is related on the assumption that the motor side reading from the sense on the Ankles joints is assumed equal to the motor ones.
This is not true, so we decided to assume that in the joint "world", motor side and link side position are equal (calculated with chengxu FK as a transmission plugin).
 
Comments are more than welcome from @alaurenzi and @ntsagarakis: if you find this not correct please open another issue on https://github.com/ADVRHumanoids/cogimon_tools

We solved the issue with @pepperig and @EnricoMingo.

alaurenzi, 2018-04-03:
I believe this is the right solution. It should be possible though to compute the joint space motor side from the motor positions before the trasmission applying chengxu ik

liesrock, 2018-04-04:
The issue was solved only partially with the above solution: @pepperig solved also the NRT part (https://github.com/ADVRHumanoids/RobotInterfaceXBotRT/commit/363eb8de0c08a4c5888df3f13fe5a33f148cbd6a), related to the robot control using the XBotCommunicationPlugin.
Remember to use valgrind in the future.

closed 2018-04-04 by liesrock