ADVRHumanoids/XBotCore #41: Control mode implementation status

alaurenzi, 2018-04-04:
**WHAT**
Control modes are a way to automatically filter/mask the reference signals that are sent to the boards. Thanks to this feature, the user can select during a startup phase which references are sent to each motor.
Example: 
`robot->setControlMode(ControlMode::Position() + ControlMode::Velocity());`
makes sure that **only** position and velocity references are sent to the robot, even if the user calls `robot->setEffortReference(des_torque);`.

**WHY**
Our robot are super complex, they are made by many joints and some of them have limitations. Example:
- centauro wrists has problems in impedance mode, so we don't want to send stiffness damping and torque to its joints
- centauro wheels are velocity-controlled, so we only want to send velocity refs to it
- controlling different parts of the robot from different ros nodes

**IMPLEMENTATION STATUS**
Until yesterday we only had getters and setters for the control mode, both joint-wise and robot-wise. Setting these quantities did not result in any effect. By pulling latest: **XBotInterface (master) - XBotCore (devel) - RobotInterfaceROS (devel) - RobotInterfaceXBotRT (master)** the user can get a beta version with the following features:
- added a **ctrl_mode** field to our `XBotCore/CommandAdvr` message. It is an unsigned 8-bit integer that represents a bit mask with the following ordering: 
    **[Unused Unused Unused Damp Stiff Effort Vel Pos]**
- failing to provide the control mode to all joints does not permit to control the robot, and a warning is printed out
- references that are filtered out by the control mode mask are not sent to the robot (filtering done inside the interface to CommandAdvr)
- RobotInterfaceROS automatically fills the message properly, from the control modes specified by the user with `RobotInterface::setControlMode()`
- RobotInterfaceXBotRT now allows to change control mode, and it should be handled properly

@EnricoMingo  @liesrock  @pepperig

liesrock, 2018-04-11:
Ok, I agree on the policy, what I would stress out is that the `RobotInterface::setControlMode()` is not commanding the actual control mode on the robot joint, but just masking some of the reference.
The encapsulation of the *ctrl_mode* inside the ROS message is a nice feature, considering it a one-shot masking for the specific message.
I am going to test it and in case opening a pull request.

liesrock, 2018-04-11:
Tested successfully in simulation (both from RT and non-RT) using RViZ and Gazebo: I just suggest to have '31' has default *ctrl_mode* instead of 0 in the ROS message.
I will open the pull request and ask to add the documentation in the XBotCore Wiki!

liesrock, 2018-04-17:
Documentation and small modification are still missing, I will close this after.