ADVRHumanoids/XBotCore #45: Getting rid of core.yaml

liesrock, 2018-04-17:
To reduce the number of config files in the system (and at the same time a bit the flexibility of the framework), me @alaurenzi and @pepperig decided to finally get rid of the core.yaml config file based also on the users feedbacks (thanks @EnricoMingo and @Pouya-moh).

This issue is related to https://github.com/ADVRHumanoids/XBotCore/issues/33 and https://github.com/ADVRHumanoids/XBotInterface/issues/45

liesrock, 2018-04-17:
The plan is to sacrifice a bit the flexibility when naming a RobotInterface, ModelInterface  or JointState / Command ROS messages.

The name policy will be the following given a NAME:

```
RobotInterfaceNAME

ModelInterfaceNAME

ROSJointStateNAME

ROSCommandNAME
```

So for example the ROS implementation of a RobotInterface will have the following name:

```
RobotInterfaceROS
```

and its related shared library will be:

```
libRobotInterfaceROS.so
```

The XBot framework is going to find the shared library specified by the user in the paths defined in the env variable *$XBOT_PATH*

Pouya-moh, 2018-04-17:
Just a suggestion, here. Maybe you could consider a very *Linuxy* practice. If you want to get dirty with getting environment variables from OS, you could consider `$HOME/.config/xbot/` as the default path for configuration file. This way you could have `core` and `interface` subdirectories with their own stuff. Just a thought...

Thanks a lot for the effort and cheers!

liesrock, 2018-06-11:
Thanks @Pouya-moh: with @alaurenzi work on https://github.com/ADVRHumanoids/XBotInterface/issues/46 and my merge/addition in devel we are able to avoid the core.yaml
Once this will be fully tested in master I will close this issue.

```
ROSJointStateNAME
ROSCommandNAME
```
Are not yet implemented: the standard is now AdvrJointStateMessage and AdvrCommandMessage

Pouya-moh, 2018-06-12:
Awesome. I have Jenkins running at all times. I can easily test it as soon as you are done.

Cheers.