ADVRHumanoids/XBotCore #54: Setup issues

williamalexmoran, 2018-06-19:
We are trying to install XBotCore for use with a humanoid robot. We followed the installation instructions for advr-superbuild and are having some issues. Every subdirectory we set to "ON" returns "Package not found. Will be downloaded and built", and when we look through the folders they are all empty. As an alternative, we also tried to clone this repository directly, but neither method found "generate_XBot_Plugin.sh". We are stuck on this step, any suggestions on how to move forward? Thanks.

pepperig, 2018-06-22:
Try to update the superbuild, then delete the build folder and try again.
Do not forget to source the terminal at the end.

pepperig, 2018-06-22:
Also this type of issue should be opened here https://github.com/ADVRHumanoids/advr-superbuild/issues

liesrock, 2018-06-24:
continuing on https://github.com/ADVRHumanoids/advr-superbuild/issues

closed 2018-06-24 by liesrock