ADVRHumanoids/XBotCore #55: Wrong config file to low level devel branch

pepperig, 2018-06-25:
I noticed that if a chose a high level config from the advr superbuild the system takes  the low level config according to the superbuild we are sourcing. So I tried to use a config from advr superbuild while I was sourcing the devel and was not working.

liesrock, 2018-06-26:
Hi @pepperig , sorry  I did not get what you are reporting: can you explain better?

pepperig, 2018-06-26:
The path to the low level config file is relative to the superbuild we source. So, if you have two superbuild you are obligated to use the config folder of the sourced superbuild. For example you cannot source the devel superbuild and use the config folder of the advr superbuild

liesrock, 2018-06-26:
This is the expected behavior: the framework needs to work only with respect to the sourced superbuild for relative paths, since it (rightly) does not know about other superbuilds

pepperig, 2018-06-26:
If it is the expected behavior it should be consistent I guess. So both high and low level config from the same place. Anyway I expect that the path inside the config would be relative to the location of the high level config. What if I have an external folder for the all the config? It won't work of I understand correctly.

liesrock, 2018-06-26:
So it is possible to set the high level config file only by its absolute path using the set_xbot_config command.

Inside fhe high level config file you can specify absolute path or relative path with respect the the $ROBOTOLOGY_ROOT set by the sourced superbuild.

So there is not any meaning for the "relative to the location of the high level config file", since we only know about the root path of the superbuild sourced and the absolute path of the high level config file.

If you have an external folder outside the superbuild you shoul currently put the absolute path in your high level config file (that can stay wherever you want).

I am working to have an XBOT_ROOT env variable to specify the location of the relative path the user want to work with: but again the relative path is always related to the config files specified inside the high level config file; the high level config file is set by the user using the set_xbot_config that will store somewhere its absolute path.

pepperig, 2018-06-26:
Ok! I thought the path was relative to the folder. So this is the expected behaviour and not an issue.

closed 2018-06-26 by pepperig