ADVRHumanoids/XBotCore #58: XBotCore: command not found

williamalexmoran, 2018-06-28:
Hello all,
I recently got XBotCore successfully installed, with it recognizing roscore, Gazebo and at first XBotCore commands. However, as soon as I opened a new terminal, it stopped recognizing XBotCore as a valid command. Are there any potential reasons why this might occur?

EnricoMingo, 2018-06-28:
Are you using the superbuild? Did you source it?

williamalexmoran, 2018-06-28:
Yes to both.

EnricoMingo, 2018-06-28:
This is very strange.
Does echo $ROBOTOLOGY_ROOT or echo $XBOT_ROOT return something?

williamalexmoran, 2018-06-28:
Those both return blank lines.

EnricoMingo, 2018-06-28:
I think the superbuild is not sourced in that terminal. Can you try to
source advr-superbuild/robotology-setup.bash and echo again?

williamalexmoran, 2018-06-28:
That seems to fix the problem where it does not recognize XBotCore as a command! It appears to be a problem with when you run XBotCore without roscore already running. It throws an error, and then XBotCore is no longer recognized. Sourcing it does fix the problem though! Thank you very much!

liesrock, 2018-06-28:
Gret! I suggest to carefully read the sections in the wiki here to help you with this kind of problems: https://github.com/ADVRHumanoids/XBotCore/wiki

liesrock, 2018-07-02:
Hi @williamalexmoran if you agree, I would close this issue.

closed 2018-07-15 by liesrock