ADVRHumanoids/XBotCore #68: XBotCore freezes during plugin initialization due to custom ros message publishers

rsubburaman, 2018-09-01:
I was trying to run a plugin which has two custom ros message publishers publishing some data to a ros topic. The plugin gets compiled without any issues and I also checked during compilation if it finds the location of the custom messages, and it does. The problem occurs during the initialization of this plugin when I start the XBotCore.

----------- In the header file, I declare the publishers as follows: 
#include <fall_custom_comm/RvizMarkersSep.h>
#include <fall_custom_comm/rviz_StateVis.h>

XBot::RosUtils::PublisherWrapper::Ptr _pub_robState, _pub_rvizData;

-------- In the source file, i define them as follows: 
_pub_robState = handle->getRosHandle()->advertise<fall_custom_comm::robot_state>("/cogimon/state", 1);
    _pub_rvizData = handle->getRosHandle()->advertise<fall_custom_comm::rviz_StateVis>("/cogimon/state/rviz", 1);

I followed the wiki example on how to define the ros publishers in the plugins. The plugin works fine with the simulator and I am facing this problem only with the real system. I couldn't find any plugin modules which actually uses a custom ros message publisher. Let me know what could be the issue or if I am missing something.

liesrock, 2018-09-01:
Ciao Raj!

Are you sure you are enabling the Xeno flag on your plugin?

Can you link here the repo of the plugin?

rsubburaman, 2018-09-01:
I guess so, because i checked the CMakeLists.txt file and it is ON by default. I have shared the repo with you, i hope you can access it. The branch which I am using on the real system is "master_cogimon_sys".

liesrock, 2018-09-05:
@rsubburaman please give us new update this

rsubburaman, 2018-09-05:
The issue was due to Xenomai being "OFF" when the XBotcore was started. @pepperig fixed the issue by turning it ON manually. This shouldn't have occurred since the Xenomai flag is ON by default in the CMakeLists.txt, looks like something else had overwritten the setting.

closed 2018-09-05 by rsubburaman